waiwnf/pilotguru #31: Have you forgotten height(altitude) in GPS record ?

QMonteCarlo, 2017-10-27:
In your GPS record in location.json file, only latitude and longitude appeared, to completely fix the position in Cartesian coordinate, isn't the altitude also necessary?

waiwnf, 2017-10-27:
Good point, thanks for pointing that out. It is definitely possible to log the altitude also, I should add that for completeness. The caveat though is that AFAIK GPS altitude is typically not as precise as horizontal coordinates, so not sure how useful that would be in practice. For my use case, I ended up only using the GPS-derived speed in the postprocessing pipeline and ignoring the coordinates altogether, but can definitely see altitude being useful in other use cases.

QMonteCarlo, 2017-10-28:
I collect these data for computer vision research purpose, specifically, for Visual Inertial SLAM. SLAM could generate the device trajectory, which could be used to compare with the GPS trajectory if available. Without altitude, such a comparison would be impossible -- even we know the altitude is not precise, we need to a way to know the range of the location error. I think it is included in the location uncertainty. 
BTW, thanks for your excellent work, I can now collect visual inertial synchronized data now anytime I want,  although I do have a visual inertial stereo camera to collect data with my laptop, but no GPS. With smartphone it is far more convenient to collect data, I found myself couldn't get away with this APP recently.

waiwnf, 2017-10-29:
Cool, good to hear this is actually useful for someone! Added altitude logging, though still not sure how precise that is. The new version should roll out on play store in the next couple of hours.
BTW the regular location accuracy in Android Location only refers to the horizontal component. There is a separate vertical accuracy component, which is also logged now, but that is only available starting with Android O.

waiwnf, 2017-10-29:
2dd83d4a5c287a3a3f22ead8c52dcf8c2c07f99b
5bb242b29bab4b726e06339f8737463b04409335

closed 2017-10-29 by waiwnf