waiwnf/pilotguru #35: Can you add barometer record?

QMonteCarlo, 2018-01-10:
Hi, waiwnf,we have talked about the quality of GPS altitude, generally it's quite unreliable. A way to work around is to use the barometer as height indicator, 1 Hz output frequency is good enough. How would you like to consider adding this to the next version?

waiwnf, 2018-01-10:
Hello. Sure, I should be able to get to it in a day or two.

waiwnf, 2018-01-13:
692d21f69b4d1d1500a9f0a323d3df7acc76c51f

waiwnf, 2018-01-13:
I think this works, but will only have time to check more thoroughly and push to play store tomorrow evening.

waiwnf, 2018-01-15:
b606cd21262d8336037d880dbff11d3fe31dab61

waiwnf, 2018-01-18:
2a7c75d3c981579b6ca86b0c70ea138d3d8ef160

New version just got pushed to play store, so closing this one. Pressure logging is off by default, you'll have to set the checkbox in preferences for it to become active. Let me know if there are any issues.

closed 2018-01-18 by waiwnf