waiwnf/pilotguru #36: visual slam tracks aligned with GPS

QMonteCarlo, 2018-01-18:
I have use this app to take records, and generate tracks from the video, then align it with the GPS track. The typical error is about 10m. The altitude from GPS is most unreliable: even when I walk on a plane, the altitude fluctuates some 20 meters. 
I'll post the aligned tracks later, if someone wants.

QMonteCarlo, 2018-09-06:
Recently I have changed my Android phone to Xiaomi 8, the first dual frequency GNSS smartphone.
The gps data quality is much better now, compared with previous old ones, however, position fluctuate still exist - after all, this is a consumer grade device.
Here is the visual trajectory aligned with gps, I have chosen to walk along known ground straight lines as a crude groundtruth.
For small area, visual slam with loop closing is more accurate.
[image]

altitude fluctuatation:
[image]

waiwnf, 2018-10-09:
Interesting, thanks for posting these! I am actually very impressed how well the optical SLAM has performed here. For my purposes (outdoors, challenging lighting, potentially quick motions) I never was able to get it to work robustly, so ended up giving up on SLAM in favor of other approaches, but great to see it working well for other use cases!

QMonteCarlo, 2018-11-10:
you're right, visual slam is not so reliable , and that's what I am working on.