waiwnf/pilotguru #37: Video record on some smartphones not focus well

QMonteCarlo, 2018-01-18:
I have installed this app on several Android phones, including Xiaomi 6, Huawei P10, Sumsung Galaxy S7.
The problem with Xiaomi 6 is that the video quality is very bad, I think the problem is about autofocus. Have you explicitly set the focus mode to "Auto focus continuous video" ? While with other android phones the video quality is quite well. The problem with Sumsung S7 is that it constantly crashes.

waiwnf, 2018-01-18:
Hello. Yeah, I was experimenting with focus a bit. The new version (will push to playstore today) will have switches for autofocus and image stabilization, so you will be able to switch continuous autofocus on in the preferences (see here for exact settings: https://github.com/waiwnf/pilotguru/blob/edafca5489139c8bb80851b8f1ebdc0aae85c0f8/mobile/android/app/src/main/java/ru/pilotguru/recorder/CaptureSettings.java#L84)

Hopefully that will resolve the quality issue.

Another thing that may be happening is the camera may have worse quality on lower resolution videos. I am using a Huawei Honor 8, and 480p videos from it were very blurry. After I switched to 720p, the quality improved a lot, so that may be something to try out. Outdoors enabling autofocus did not change much. I guess everything is far enough away that constant focus on infinity works well. Not sure about indoors. Also autofocus + digital image stabilization simultaneously is usually a bad idea, since the frame rate becomes very unsteady.

I will try to run in an S7 emulator when I have time, though will probably take a few days. Which exact android version you are using there that crashes?

waiwnf, 2018-01-18:
2a7c75d3c981579b6ca86b0c70ea138d3d8ef160

The new play store version has autofocus and image stabilization toggles.

QMonteCarlo, 2018-01-20:
Hi, I have installed the new version, and it works perfectly, thanks, I'm so happy with it.
The S7 os is android 7.0, it's from my friend, I don't use it frequently, but I'll try the new version on it.

waiwnf, 2018-01-24:
Cool, good to know things work! Let me close this one and make a separate bug for S7 crashes..

closed 2018-01-24 by waiwnf