waiwnf/pilotguru #43: Could gps recording add the number of satellites observed?

QMonteCarlo, 2020-03-18:
Hi! I've been enjoying this app for two years and this app also improved a lot. 
And I found one issue relevant to the robustness of location algorithm:  the GPS recording lacks the number of satellites observed , which could be read from GpsSvStatus.num_svs. This is important when you go from outdoor into a building, for a location algorithm we need to know that GPS should not be used anymore by the number of satellites (num_svs), but the app still gives gps recordings and the accuracy is no changed, certainly an algorithm would be misled by such recordings. Would it be possible to add this recording in the future release versions?

waiwnf, 2020-03-18:
Hi Yi. Hm, I have not really tested indoors much and just assumed that Android would take care of that by changing accuracy. I'll try to look into this in a day or two.

QMonteCarlo, 2020-03-19:
I'd like to share my test result. 
For my experience, the GPS recording ended seconds later when I was indoor , this is a good news. But for some cases, the transition from outdoor to indoor is not so clear-cut, the GPS uncertainty did not increase accordingly, for example, when I was actually going to underground while satellite signal is still vaguely observable, the altitude increased, and the uncertainty does not change much.
In a world, I believe a recording of GPS integration is helpful.

waiwnf, 2020-03-19:
893ac0c9188f3974a15de72b2b599143f10e7f68

waiwnf, 2020-03-19:
Hi Yi. I have pushed a new release that logs the number of active satellites to a separate file (`gps_status.json`). Please let me know if there are any issues once you get the update.

QMonteCarlo, 2020-03-19:
Hi, I'm not able to compile it my self(it's an Android app,right?) until it is in Googleplay or any other app store available.
Besides, I just wonder, is the speed_m_s the norm of the total speed vector, and bearing degree is the azimuth in local ENU? Whether the accuracy is actually the horizontal accuracy or total accuracy(that means, vertical accuracy is part of it)?

waiwnf, 2020-03-20:
Hi Yi. I have pushed an update to the play store as well, but it looks like it will take some time (it says up to a week in the interface) to get reviewed. In the meantime, you can compile it yourself - clone the repo, install android studio (https://developer.android.com/studio) and open mobile/android subdirectory in android studio. It should recognize a project and let you build the app.

The speed and bearing I honestly have not actually used. They are logged straight from the `Location` object, and the docs have some info: [](https://developer.android.com/reference/android/location/Location#getSpeed()) and [](https://developer.android.com/reference/android/location/Location#getBearing()). It does not even say what zero bearing corresponds to, but looking at [](https://developer.android.com/reference/android/location/Location#bearingTo(android.location.Location)) I assume it is the same - degrees East wrt the true North.

The accuracy field is horizontal component only (per [](https://developer.android.com/reference/android/location/Location#getAccuracy())). On newer Androids it should also log vertical accuracy component as a separate field (see [](https://github.com/waiwnf/pilotguru/blob/893ac0c9188f3974a15de72b2b599143f10e7f68/mobile/android/app/src/main/java/ru/pilotguru/recorder/SensorDataSaver.java#L287)).

QMonteCarlo, 2020-03-24:
Thanks, I'll try as soon as I can.

closed 2020-03-24 by QMonteCarlo